ros-industrial-attic/ur_modern_driver #345: Question: minimum requirements

gonzalocasas, 2019-09-01:
Considering the `ur_modern_driver` is so demanding on the computer and network to work reliably, has anyone determined & documented its minimum requirements already? (When not using the low bandwidth trajectory follower).

Alternatively -or perhaps even better than a static requirements list-, a start-up test that verifies that the system can sustain latency and bandwidth required might go a really long way at reducing (new) user's frustration.

gavanderhoorn, 2019-09-04:
No, the minimum requirements have not been determined as far as I know.

It's also going to depend quite a bit on system utilisation: an RPi with no other tasks than the driver can run it quite successfully, but an i7 with 32GB of ram running 30+ nodes and Gazebo may have trouble sustaining the required rates. And that is disregarding factors like network bandwidth, contention on the segment, etc.

Note: I'm not claiming this would not be good information to have. Just noting that it's not a trivial thing to do/measure.

gavanderhoorn, 2019-09-04:
#321 could help, but for that to *really* work, users would have to install an RT kernel, which will probably not help make things "easier" for the average user.

gavanderhoorn, 2019-11-23:
By closing this, I'm not dismissing it, but I feel it would make sense to try and derive some system requirements for `ur_robot_driver` instead.

If you'd agree, please open an issue on the `ur_robot_driver` tracker to discuss.

closed 2019-11-23 by gavanderhoorn