ros-industrial-attic/ur_modern_driver #346: UR5e doesnt simulate

addy1997, 2019-09-04:
I am getting these errors in my terminal.

[ERROR] [1567586275.334899795]: Robot model parameter not found! Did you remap 'robot_description'?
[ERROR] [1567586275.405380317]: Robot model not loaded
[ERROR] [1567586275.435468528]: Planning scene not configured

gavanderhoorn, 2019-09-04:
This is not enough information to be able to help you.

You're going to have to provide at least all the commands that you use to start things up, and tell us what it is that you're actually trying to do.

As a first question: which simulation are you referring to? The basic MoveIt `demo.launch`, or with Gazebo?

addy1997, 2019-09-07:
ohk

addy1997, 2019-09-11:
I have made a code for circular path trajectory want to get it checked. Please reply immediately ASAP!!!!

def Circular_path(self,pos_x,pos_y,pos_z,waypoint,radius):
        group = self.group
        scene = self.scene
        eef_link = self.eef_link
        robot = self.robot
        radius = self.radius
        waypoints = self.waypoints
        
        x_center, y_center = input("Enter a two value: ").split() 
        print("X Coordinate is: ", x_center) 
        print("Y Coordiante is: ", y_center) 
        print() 
        
        # Defining trajectory parameters. 
        double angle1 = angle_resolution*3.14/180
        double angle = 0
        
        # Main logic for circular trajectory.
        for i in range(0, 360/angle_resolution)
        angle+=angle1
        eef_link.pos_x = x_center + r*cos(angle)
        eef_link.pos_y = y_center + r*sin(angle)
        eef_link.pos_z = 0 # because circle is 2-d i.e, we have to follow the trajectory of the cross-section of the sphere.
        
        waypoints.append(copy.deepcopy(eef_link))

        plan = move_group.compute_cartesian_path(
                                   waypoints,   # waypoints to follow  
                                   0.0) 
    
        
        display_trajectory = moveit_msgs.msg.Circular_path()
        display_trajectory.trajectory_start = robot.get_current_state()
        display_trajectory.trajectory.append(plan)
     
        # Publish
        display_trajectory_publisher.publish(display_trajectory)
     
        # To Execute the Planned Path 
        move_group.execute(plan, wait=True)

    
        print('The robot moves in a circular path')

closed 2019-09-11 by addy1997

gonzalocasas, 2019-09-11:
> I have made a code for circular path trajectory want to get it checked.
> Please reply immediately ASAP!!!!

Please, be more thoughtful/polite on these channels. The all uppercase, multiple exclamation mark, double urgency qualifiers will not really get anything done faster.

Also, you're posting code that is: a) incomplete (the method belongs to a class but it's not included b) incorrectly formatted (please use triple quote to wrap the entirety of it, not just a half) c) in the wrong place: GitHub issues are not to request reviews, that's what pull requests are for.

Please review the contribution guidelines beforehand, post a pull request, and people will be much more likely to help you out.

addy1997, 2019-09-11:
I apologise for the misconduct. It's not in slightest of my intension to
break the open source guidelines.

Please Sir, would you help me debug my code as I need it urgently for my
project.
I am attaching a file python file with full code for your reference.

I regret for wasting your time.
Thank you
Adwait

gavanderhoorn, 2019-09-11:
I'm afraid we don't review code submissions from users here on this tracker, as @gonzalocasas also writes.

Submitting it as part of a PR would be more appropriate, but in this case the code appears to be completely UR-agnostic (in fact, it would seem to be robot-agnostic), which makes it out-of-scope for this repository.

I would suggest to post a question on ROS Answers, as that is a more general support forum.

However, please keep in mind that even ROS Answers does not run a code review service.