ros-industrial-attic/ur_modern_driver #347: Can't connect two ur10 by ur10_ros_control.launch

zhengmaoch, 2019-09-09:
Hi, the follow is my controller file and launch file. 

#hnrob_left_arm_controller

`# Settings for ros_control control loop
hardware_control_loop:
   loop_hz: 125

# Settings for ros_control hardware interface
hardware_interface:
   joints:
     - left_shoulder_pan_joint
     - left_shoulder_lift_joint
     - left_elbow_joint
     - left_wrist_1_joint
     - left_wrist_2_joint
     - left_wrist_3_joint

# Publish all joint states ----------------------------------
left_joint_state_controller:
   type:         joint_state_controller/JointStateController
   publish_rate: 125

# Publish wrench ----------------------------------
left_force_torque_sensor_controller:
   type:         force_torque_sensor_controller/ForceTorqueSensorController
   publish_rate: 125

# Joint Trajectory Controller - position based -------------------------------
# For detailed explanations of parameter see http://wiki.ros.org/joint_trajectory_controller
left_pos_based_pos_traj_controller:
   type: position_controllers/JointTrajectoryController
   joints:
     - left_shoulder_pan_joint
     - left_shoulder_lift_joint
     - left_elbow_joint
     - left_wrist_1_joint
     - left_wrist_2_joint
     - left_wrist_3_joint

   # state_publish_rate:  50 # Defaults to 50
   # action_monitor_rate: 20 # Defaults to 20
   #stop_trajectory_duration: 0 # Defaults to 0.0

# Joint Trajectory Controller -------------------------------
# For detailed explanations of parameter see http://wiki.ros.org/joint_trajectory_controller
left_vel_based_pos_traj_controller:
   type: velocity_controllers/JointTrajectoryController
   joints:
     - left_shoulder_pan_joint
     - left_shoulder_lift_joint
     - left_elbow_joint
     - left_wrist_1_joint
     - left_wrist_2_joint
     - left_wrist_3_joint
   gains:
      #!!These values have not been optimized!!
      left_shoulder_pan_joint:  {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      left_shoulder_lift_joint: {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      left_elbow_joint:         {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      left_wrist_1_joint:       {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      left_wrist_2_joint:       {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      left_wrist_3_joint:       {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}

   # Use a feedforward term to reduce the size of PID gains
   velocity_ff:
      left_shoulder_pan_joint: 1.0
      left_shoulder_lift_joint: 1.0
      left_elbow_joint: 1.0
      left_wrist_1_joint: 1.0
[... 408 lines cut ...]
log file: /home/fenjin/.ros/log/a877df70-d2db-11e9-86db-3c6aa740d0df/right_ros_control_controller_manager-3*.log
[INFO] [1568017712.451550]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1568017712.452922]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1568017712.454316]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1568017712.455703]: Loading controller: right_joint_state_controller
[INFO] [1568017712.462197]: Loading controller: right_pos_based_pos_traj_controller
[ERROR] [1568017713.470709]: Failed to load right_pos_based_pos_traj_controller
[INFO] [1568017713.471101]: Controller Spawner: Loaded controllers: right_joint_state_controller
[ERROR] [1568017713.472964314]: Could not start controller with name right_joint_state_controller because no controller with this name exists
[ERROR] [1568017713.473204]: Failed to start controllers: right_joint_state_controller`

#terminal1
`
[ERROR] [1568017712.299555223]: Exception thrown: Could not find resource 'right_shoulder_pan_joint' in 'VelocityInterface'.
[ERROR] [1568017712.299598315]: Failed to initialize the controller
[ERROR] [1568017712.299623155]: Initializing controller 'right_joint_group_vel_controller' failed
[ERROR] [1568017712.310559036]: Could not find joint 'right_shoulder_pan_joint' in 'hardware_interface::VelocityJointInterface'.
[ERROR] [1568017712.310614303]: Failed to initialize the controller
[ERROR] [1568017712.310631314]: Initializing controller 'right_vel_based_pos_traj_controller' failed
[ERROR] [1568017712.469722644]: Could not find joint 'right_shoulder_pan_joint' in 'hardware_interface::PositionJointInterface'.
[ERROR] [1568017712.469788770]: Failed to initialize the controller
[ERROR] [1568017712.469819240]: Initializing controller 'right_pos_based_pos_traj_controller' failed
[left_arm_hardware_interface-2] process has died [pid 5173, exit code 1, cmd /home/fenjin/hnrob_ws/devel/lib/ur_modern_driver/ur_driver __name:=left_arm_hardware_interface __log:=
/home/fenjin/.ros/log/a877df70-d2db-11e9-86db-3c6aa740d0df/left_arm_hardware_interface-2.log].
log file: /home/fenjin/.ros/log/a877df70-d2db-11e9-86db-3c6aa740d0df/left_arm_hardware_interface-2*.log`

gavanderhoorn, 2019-09-09:
This will only work if you also have a composite xacro that combines two instances of the `ur10_robot` xacro macro with the appropriate `left_` and `right_` prefixes.

Or two stand-alone `ur10_robot` instances (ie: not in the same file/model) still with the appropriate prefixes *and* two `robot_description` parameters in the appropriate namespace.

If you don't have that, that would explain why the controllers cannot find the joints (as there are no `right_...` and `left_...` joints in the default xacro macro).

gavanderhoorn, 2019-09-09:
As this is a configuration issue and not an issue with the package in this repository, I'm going to close this.

If it turns out this is indeed an issue with `ur_modern_driver` (and not with the configuration, `ros_control` controllers or something else), we can re-open and investigate.

In the meantime, please feel free to keep commenting of course.

closed 2019-09-09 by gavanderhoorn

zhengmaoch, 2019-09-09:
yes, I hava a composite urdf that combines two instances, follow is my urdf.

`Hi, the follow is my controller file and launch file. 

#hnrob_left_arm_controller

`# Settings for ros_control control loop
hardware_control_loop:
   loop_hz: 125

# Settings for ros_control hardware interface
hardware_interface:
   joints:
     - left_shoulder_pan_joint
     - left_shoulder_lift_joint
     - left_elbow_joint
     - left_wrist_1_joint
     - left_wrist_2_joint
     - left_wrist_3_joint

# Publish all joint states ----------------------------------
left_joint_state_controller:
   type:         joint_state_controller/JointStateController
   publish_rate: 125

# Publish wrench ----------------------------------
left_force_torque_sensor_controller:
   type:         force_torque_sensor_controller/ForceTorqueSensorController
   publish_rate: 125

# Joint Trajectory Controller - position based -------------------------------
# For detailed explanations of parameter see http://wiki.ros.org/joint_trajectory_controller
left_pos_based_pos_traj_controller:
   type: position_controllers/JointTrajectoryController
   joints:
     - left_shoulder_pan_joint
     - left_shoulder_lift_joint
     - left_elbow_joint
     - left_wrist_1_joint
     - left_wrist_2_joint
     - left_wrist_3_joint

   # state_publish_rate:  50 # Defaults to 50
   # action_monitor_rate: 20 # Defaults to 20
   #stop_trajectory_duration: 0 # Defaults to 0.0

# Joint Trajectory Controller -------------------------------
# For detailed explanations of parameter see http://wiki.ros.org/joint_trajectory_controller
left_vel_based_pos_traj_controller:
   type: velocity_controllers/JointTrajectoryController
   joints:
     - left_shoulder_pan_joint
     - left_shoulder_lift_joint
     - left_elbow_joint
     - left_wrist_1_joint
     - left_wrist_2_joint
     - left_wrist_3_joint
   gains:
      #!!These values have not been optimized!!
      left_shoulder_pan_joint:  {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      left_shoulder_lift_joint: {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      left_elbow_joint:         {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      left_wrist_1_joint:       {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      left_wrist_2_joint:       {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}
      left_wrist_3_joint:       {p: 5.0,  i: 0.05, d: 0.1, i_clamp: 1}

   # Use a feedforward term to reduce the size of PID gains
   velocity_ff:
      left_shoulder_pan_joint: 1.0
      left_shoulder_lift_joint: 1.0
[... 1361 lines cut ...]
                <resolution>1</resolution>
                <min_angle>-1.570796</min_angle>
                <max_angle>1.570796</max_angle>
              </horizontal>
            </scan>
            <range>
              <min>0.10</min>
              <max>30.0</max>
              <resolution>0.01</resolution>
            </range>
            <noise>
              <type>gaussian</type>
              
              <mean>0.0</mean>
              <stddev>0.01</stddev>
            </noise>
          </ray>
          <plugin name="gazebo_ros_head_hokuyo_controller" filename="libgazebo_ros_laser.so">
            <topicName>/hnrob/laser/scan</topicName>
            <frameName>hokuyo_link</frameName>
          </plugin>
        </sensor>
    </gazebo>
    
</robot>`

UmutUz, 2021-02-23:
Does anyone ever solved this problem? I try practically the same thing: Have 2 robots UR10, have one xacro for each robot and call them from a third one. Problem is: When launching the bringup the joints of one robot are not found (same error as the OP). Ive wasted days trying to get the controllers work. Does anyone have an idea how to do the controller files?

gavanderhoorn, 2021-02-23:
@UmutUz: can you tell us what controller you have? A CB3 or a CB2?

UmutUz, 2021-02-23:
@gavanderhoorn I use a CB3. ROS Melodic Ubuntu 18.04. Tried lots of lots of stuff. Tried to replicate the controller.yaml made by the moveit setup assistant, namespaces, groups thought I would try to load 2 yamls and managed to get as far as OP but now I am stuck same error over and over. :/

UmutUz, 2021-02-23:
@gavanderhoorn I used pretty much everything from the Universal_Robots_Driver git, working fine with one robot, making the urdf of 2 worked fine in rviz too (using the fake_controllers.yaml). I have tried so many things at this point I really don't know what else I could do lol.

gavanderhoorn, 2021-02-23:
> I use a CB3.

then please do not post here, as this driver should not be used with CB3 controllers any more.

> I used pretty much everything from the Universal_Robots_Driver git,

Then please post your issue on the tracker there, instead of in a comment under an already closed issue on a tracker of a deprecated driver.

UmutUz, 2021-02-23:
@gavanderhoorn alright, thanks for guiding!