ros-industrial-attic/ur_modern_driver #350: Compatibility with UR10e software version 5.5.1

Drikuy, 2019-09-18:
Hi everyone!

I've received this moorning an UR e-mail that tolds me to update into software version 5.5.1. I am wondering if this controller version has compatibility with actual `ur_modern_driver`. When I update into this versión and test connection with ROS I've get the `Sub-package of type 3 was not parsed completely` error message and the ur_driver node shuts down.

I've tried to downgrade the versión but I've still get the same error as you can see:

```
[ INFO] [1568796249.990101834]: Setting up connection: 172.16.7.75:30003
[ INFO] [1568796249.993175427]: Connection established for 172.16.7.75:30003
[ INFO] [1568796249.993260537]: Setting up connection: 172.16.7.75:30001
[ INFO] [1568796249.994485808]: Connection established for 172.16.7.75:30001
[ INFO] [1568796250.317234269]: Got VersionMessage:
[ INFO] [1568796250.317274660]: project name: URControl
[ INFO] [1568796250.317284556]: version: 5.4.0
[ INFO] [1568796250.317293127]: build date: 21-06-2019, 10:03:02
[ INFO] [1568796250.317301908]: Disconnecting from 172.16.7.75:30001
[ INFO] [1568796250.334145295]: ActionServer enabled
[ INFO] [1568796250.334217296]: Use standard trajectory follower
[ INFO] [1568796250.337357074]: Setting up connection: :50001
[ INFO] [1568796250.337518474]: Connection established for :50001
[ INFO] [1568796250.337668684]: Initializing ur_driver/URScript subscriber
[ INFO] [1568796250.344516354]: The ur_driver/URScript initialized
[ INFO] [1568796250.344603230]: Notifier: Pipeline disconnect will shutdown the node
[ INFO] [1568796250.359143363]: Service 'ur_driver/robot_enable' activation mode: Never
[ INFO] [1568796250.359287744]: Starting main loop
[ INFO] [1568796250.359841096]: Setting up connection: 172.16.7.75:30003
[ INFO] [1568796250.360033299]: Starting pipeline RTPacket
[ INFO] [1568796250.360145653]: Starting pipeline StatePacket
[ INFO] [1568796250.360344891]: Setting up connection: 172.16.7.75:30002
[ INFO] [1568796250.361160789]: Connection established for 172.16.7.75:30003
[ INFO] [1568796250.362643187]: Connection established for 172.16.7.75:30002
[ INFO] [1568796250.363747323]: Starting ActionServer
[ INFO] [1568796250.364068057]: Trajectory thread started
[ERROR] [1568796250.379413885]: Sub-package of type 3 was not parsed completely!
[ INFO] [1568796250.379490222]: Disconnecting from 172.16.7.75:30002
[ INFO] [1568796250.379661885]: Shutting down on stopped pipeline StatePacket
[ur_driver-3] process has died ...
```

Does anyone knows the answer or knows how to solve this issue? 
Thanks in advance for your help

martimorta-wood, 2019-09-23:
This is what I did to fix it, it is not clean but it works:

https://github.com/ros-industrial/ur_modern_driver/pull/216#issuecomment-493475540

Drikuy, 2019-10-06:
I'm sorry I answer so late. Thanks @martimorta-wood!

gavanderhoorn, 2019-11-23:
Please use `ur_robot_driver` for all CB3 and e-series controllers instead of `ur_modern_driver`.

`ur_modern_driver` will not be updated for improve compatibility with newer Polyscope versions.

closed 2019-11-23 by gavanderhoorn