ros-industrial-attic/ur_modern_driver #352: UR5 : Sub-package of type 0 was not parsed completely!

DentOpt, 2019-09-19:
I am using Polyscope 3.2.20175 
and URControl 3.2.20175,
unfortunately the ur_modern_driver (kinetik branch with additionally installed industrial_msgs)
does not connect to the UR5 robotic arm.
Pinging the robot on 192.168.0.101 works without issues. 

```
process[robot_state_publisher-2]: started with pid [2739]
process[ur_driver-3]: started with pid [2740]
[ INFO] [1568896339.309695815]: Setting up connection: 192.168.0.101:30003
[ INFO] [1568896339.314847808]: Connection established for 192.168.0.101:30003
[ INFO] [1568896339.314875650]: Setting up connection: 192.168.0.101:30001
[ INFO] [1568896339.316518189]: Connection established for 192.168.0.101:30001
[ INFO] [1568896339.490758137]: Got VersionMessage:
[ INFO] [1568896339.490821721]: project name: URControl
[ INFO] [1568896339.490847432]: version: 3.2.20175
[ INFO] [1568896339.490874909]: build date: 29 2016, 10:43:26
[ INFO] [1568896339.490901604]: Disconnecting from 192.168.0.101:30001
[ INFO] [1568896339.507266817]: ActionServer enabled
[ INFO] [1568896339.509690124]: Use standard trajectory follower
[ INFO] [1568896339.512902851]: Setting up connection: :50001
[ INFO] [1568896339.513249726]: Connection established for :50001
[ INFO] [1568896339.513386975]: Initializing ur_driver/URScript subscriber
[ INFO] [1568896339.519818154]: The ur_driver/URScript initialized
[ INFO] [1568896339.519887246]: Notifier: Pipeline disconnect will shutdown the node
[ INFO] [1568896339.528515583]: Service 'ur_driver/robot_enable' activation mode: Never
[ INFO] [1568896339.528606156]: Starting main loop
[ INFO] [1568896339.528917636]: Setting up connection: 192.168.0.101:30003
[ INFO] [1568896339.529103166]: Starting pipeline RTPacket
[ INFO] [1568896339.529380487]: Setting up connection: 192.168.0.101:30002
[ INFO] [1568896339.529547097]: Starting pipeline StatePacket
[ INFO] [1568896339.531457898]: Connection established for 192.168.0.101:30002
[ INFO] [1568896339.532188658]: Connection established for 192.168.0.101:30003
[ INFO] [1568896339.532946588]: Starting ActionServer
[ INFO] [1568896339.533247302]: Trajectory thread started
[ERROR] [1568896339.626585933]: Sub-package of type 0 was not parsed completely!
[ INFO] [1568896339.626716896]: Disconnecting from 192.168.0.101:30002
[ INFO] [1568896339.626874699]: Shutting down on stopped pipeline StatePacket
```

Does anyone know ho to solve the parsing issue?

DentOpt, 2019-09-20:
I have updated the UR Software to 3.11 and the error changed to:

`SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /ur_driver/base_frame: base
 * /ur_driver/max_joint_difference: 0.01
 * /ur_driver/max_payload: 5.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/max_waiting_time: 2.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix: 
 * /ur_driver/require_activation: Never
 * /ur_driver/reverse_ip_address: 
 * /ur_driver/reverse_port: 50001
 * /ur_driver/robot_ip_address: 192.168.0.102
 * /ur_driver/servoj_gain: 100.0
 * /ur_driver/servoj_lookahead_time: 1.0
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/servoj_time_waiting: 0.001
 * /ur_driver/shutdown_on_disconnect: True
 * /ur_driver/time_interval: 0.008
 * /ur_driver/tool_frame: tool0_controller
 * /ur_driver/use_lowbandwidth_trajectory_follower: False
 * /ur_driver/use_ros_control: False

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

auto-starting new master
process[master]: started with pid [14840]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e2ab309a-dbb2-11e9-96fc-9cb6d0fef5eb
process[rosout-1]: started with pid [14851]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [14858]
process[ur_driver-3]: started with pid [14859]
[ INFO] [1568989697.425398625]: Setting up connection: 192.168.0.102:30003
[ INFO] [1568989697.430230056]: Connection established for 192.168.0.102:30003
[ INFO] [1568989697.430260547]: Setting up connection: 192.168.0.102:30001
[ INFO] [1568989697.431982520]: Connection established for 192.168.0.102:30001
[ INFO] [1568989697.841148277]: Got VersionMessage:
[ INFO] [1568989697.841207261]: project name: URControl
[ INFO] [1568989697.841231282]: version: 3.11.0
[ INFO] [1568989697.841253083]: build date: 25-07-2019, 12:32:58
[ INFO] [1568989697.841280150]: Disconnecting from 192.168.0.102:30001
[ INFO] [1568989697.856877958]: ActionServer enabled
[ INFO] [1568989697.861387766]: Use standard trajectory follower
[ INFO] [1568989697.866445421]: Setting up connection: :50001
[ INFO] [1568989697.866779020]: Connection established for :50001
[ INFO] [1568989697.866910853]: Initializing ur_driver/URScript subscriber
[ INFO] [1568989697.872662725]: The ur_driver/URScript initialized
[ INFO] [1568989697.872741675]: Notifier: Pipeline disconnect will shutdown the node
[ INFO] [1568989697.881677757]: Service 'ur_driver/robot_enable' activation mode: Never
[ INFO] [1568989697.881778645]: Starting main loop
[ INFO] [1568989697.882066699]: Setting up connection: 192.168.0.102:30003
[ INFO] [1568989697.882242259]: Starting pipeline RTPacket
[ INFO] [1568989697.882489894]: Starting pipeline StatePacket
[ INFO] [1568989697.882633450]: Setting up connection: 192.168.0.102:30002
[ INFO] [1568989697.884098256]: Connection established for 192.168.0.102:30003
[ INFO] [1568989697.884609915]: Connection established for 192.168.0.102:30002
[ INFO] [1568989697.885150983]: Starting ActionServer
[ INFO] [1568989697.885436389]: Trajectory thread started
[ERROR] [1568989697.926303307]: Sub-package of type 3 was not parsed completely!
[ INFO] [1568989697.926428830]: Disconnecting from 192.168.0.102:30002
[ INFO] [1568989697.926543599]: Shutting down on stopped pipeline StatePacket
`

gavanderhoorn, 2019-09-20:
Now this would be a duplicate of #316.

gavanderhoorn, 2019-11-23:
`ur_robot_driver` does not use the primary, secondary or matlab/real-time interfaces to avoid these kinds of issues.

closed 2019-11-23 by gavanderhoorn