ros-industrial-attic/ur_modern_driver #353: The program block after released protective stop

sean85914, 2019-09-20:
UR Firmware Version: **3.9.0**.64176

**Problem**: 
I used `follow_joint_trajectory` interface to control my UR5 (refer from [here](http://wiki.ros.org/pr2_controllers/Tutorials/Moving%20the%20arm%20using%20the%20Joint%20Trajectory%20Action))
I send the first waypoint to the robot, while the obstacle make it protective stop, I enabled the robot
and keep send second watpoint, while the terminal just blocks at

> ...
[ INFO] [1568756153.361066266]: Received new goal
[ INFO] [1568756153.361118846]: Trajectory received and accepted
[ INFO] [1568756153.361153748]: Translating trajectory
[ INFO] [1568756153.361191331]: Executing trajectory with 2 points and duration of 3.000s
[ INFO] [1568756153.361210379]: Attempting to start follower 0x1a9c300
[ INFO] [1568756153.361227916]: **Uploading trajectory program to robot**

I had to manually relaunch the `ur5_bringup.launch` to keep sending waypoints to the robot.

My question is that, is that possible I keep sending waypoints to the robot after I released the protective stop without relaunch the file, or why this issue happened?

Any suggestion is very appreciated, thanks

sean85914, 2019-09-21:
Close the issue since I found out I was using old repository from Thomas

sean85914, 2019-09-21:
The issue remains after updating to the newest version from this repository

megantuttle, 2019-11-05:
I am experiencing a similar issue with a UR3 (firmware version 3.6.1.30595) using the depreciated driver.  I am using the position based controller to send commands to the robot.  Everything works as it should prior to reaching a protective stop condition.  Once the protective stop is released, the robot no longer responds to commands from the position based controller.  I added a few URScript statements in `driverProg()` to print to the log in the UR controller and it appears that `driverProg()` is hanging/no longer running.  I can switch to the velocity based controller and the robot will respond to commands.  When switching back to the position based controller, `uploadProg()` is called and the UR controller should indicate in the log that `driverProg()` has started.  But it doesn't so the program isn't running and so of course the robot can't respond to commands.  There is no error output in this case.  Has anyone else experienced this issue?  I realize I am using the depreciated driver but it isn't really an option to use the refactor at this point.

Among other things, I've tried saving `driverProg()` to the UR controller's memory and calling it from the dashboard server after releasing the protective stop; the result there is that `socket_read_binary_integer()` times out and the message: `Not able to open socket socket_0 at host: X.X.X.X at port 50001` appears in the UR controller log.

The only way to get the position controller to work again is to re-launch the driver entirely, and this is undesirable behavior.  Any insight/help on this issue would be greatly appreciated.  I'll happily post screenshots/logs, just let me know what might be useful.  Thanks!

gavanderhoorn, 2019-11-23:
If this appears to also be an issue with `ur_robot_driver`, please open a new issue on the `ur_robot_driver` tracker.

closed 2019-11-23 by gavanderhoorn