ros-industrial-attic/ur_modern_driver #354: Connecting to real UR5e robot

Ruthy85, 2019-10-01:
Hello, I am using the  'universal_robot' driver in Ubuntu Xenial with ROS Kinetic.
With the old driver, I used bringup.launch to start the communication with the real robot as:

> roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=myIP

However, there is no bringup.launch for ur5e, and other launch files haven't got the IP as an argument.
I tried to start the ur5e description launch and then start rviz, but there is no connection to real robot, so IP was not provided.
How to work with a real ur5e then? Thank you

gonzalocasas, 2019-10-02:
@Ruthy85 looks like you're correct, the launch files for E-series indeed seem to be missing. However, it should be simple to add them, could you please try to copy `ur5_bringup.launch` and replace the loading of the model with e-series' one? i.e.
replace the following:
```
  <!-- robot model -->
  <include file="$(find ur_description)/launch/ur5_upload.launch">
    <arg name="limited" value="$(arg limited)"/>
  </include>
```
with
```
  <!-- robot model -->
  <include file="$(find ur_e_description)/launch/ur5e_upload.launch">
    <arg name="limited" value="$(arg limited)"/>
  </include>
```

gonzalocasas, 2019-10-02:
Sorry, please ignore my previous comment. 

e-Series support is currently being worked on on a fork, you can check the pull request here: https://github.com/ros-industrial/ur_modern_driver/pull/339 
To test it, pull that fork into your catkin workspace, test if it works on your setup and ideally report back in the pull request.

Ruthy85, 2019-10-02:
Thank you. 
I am testing this fork, that has already the eSeries files.
However, I am getting this error when bringup the robot:

> unused args [reverse_port] for include of [/home/ros/catkin_ws/src/ur_modern_driver/launch/ur_common.launch]

Even if I pass an argument for the 'reverse_port' I've got the error. However, this argument should be optional...

Ruthy85, 2019-10-07:
I already have it running. Problem was a wrong merge to kinetic-devel.
Now I can see the robot real pose and its update when it's moved manualy.
However, I am not able to plan, because it's always Failed, due to:

```
You can start planning now!

[ INFO] [1570454211.959254809]: Planning request received for MoveGroup action. Forwarding to planning pipeline.
[ WARN] [1570454211.959419965]: Joint 'wrist_1_joint' from the starting state is outside bounds by a significant margin: [ 4.68739 ] should be in the range [ -3.14159 ], [ 3.14159 ] but the error above the ~start_state_max_bounds_error parameter (currently set to 0.1)
```

SunYuansong, 2019-10-17:
hi, I meet the similar problem when using UR3e with ur modren driver:
[ERROR] [1570010886.429175180]: Sub-package of type 0 was not parsed completely!
how do you solve this at last? thanks

Ruthy85, 2019-10-17:
Hi, I was using a wrong merge to the kinetic-devel branch. After merging to it, everything works fine.

closed 2019-10-18 by gavanderhoorn