ros-industrial-attic/ur_modern_driver #355: URScript feedback

Drikuy, 2019-10-02:
Hi everyone,

I'm working with an UR10e robot and I want to send a program to open and close a gripper through `/ur_driver/URScript`. I've already send properly the command to the robot publishing into this topic (and it works). However I want to know if `ur_driver` node publishes any answer from the robot when you send a wrapped program vía URScript command. I want to know this in order to make some waiting code lines until an answer is listened before making the next move after closing the gripper.

gavanderhoorn, 2019-10-02:
Please see whether #162 covers this. Especially the later comments.

In the meantime I'm going to close this, as it reads like a duplicate. If it isn't, we can always re-open.

closed 2019-10-02 by gavanderhoorn

Drikuy, 2019-10-03:
Thanks @gavanderhoorn ! I don't know why I cannot see here your answer but as you said, it is already answered on this other closed topic #162 .