ros-industrial-attic/ur_modern_driver #356: Real-time conrol of UR5 in force mode using ur_modern_driver

xjboston, 2019-10-02:
Hi, 
I have question regarding implementing the force_mode() function with ur_modern_driver. 
We have been doing real-time teleoperation of a UR5 arm using ur_modern_driver. In the current setup, we sent velocity commands to each joint in realtime and the test went pretty well. The only issue is that the robot is very stiff, which I think is due to the high internal loop gains. We are now finding solutions to make the robot compliant.  Here are some ideas we want to try.
1. Modify the loop gains. We have contacted with the company about this and were told that it is not feasible.
2.  Add torque sensors and implement admittance/impedance controller. The designing of an impedance controller is not easy. We want to leave it as the last option.
3. Using the built-in force_mode. The UR5 has a built-in force mode where the user can select directions in which the robot is compliant. The function was explained in their scriptMannual. We have also tested it using Polyscope. This seems to be a possible way. 
I was wondering that is there any way to implement this 'force_mode()' function into the ur_modern_driver? Based on my understanding, the real-time interface in ur_modern_driver is typically running the built-in function speedj() to read the velocity commands (defined in the file 'ur_realtime_communication.cpp'). It might be possible to do the same thing with force_mode().
Please let me know your thoughts on this and correct me if I am wrong anywhere. Any suggestions would be really appreciated. Thanks!

xjboston, 2019-10-07:
The compliance behavior can be realized by a kinetic compliance controller (https://github.com/UTNuclearRobotics/ros_controllers/tree/kinetic-compliance/compliance_controller), which is exactly what I am after.

closed 2019-10-07 by xjboston

ZZWang21, 2022-10-07:
Excuse me @xjboston  @PaperFanz @AndyZe, I am also working on the compliance controller. I am wondering, how to set a force target for this controller? Currently, I looked through the codes, but I did not see where I can set a constant force target or goal, or desired force. Thank you.

AndyZe, 2022-10-07:
Hi @ZZWang21, it's been a long time since I touched this particular code. If you can upgrade to ROS2 Humble, there's a new admittance controller that's supported better and has more features. I'd suggest using that one instead:

https://github.com/ros-controls/ros2_control_demos/pull/206

I'm 95% sure this pull request is going to be merged in the next few weeks, so it's officially available in ROS2.

ZZWang21, 2022-10-07:
Thank you, Andy, for the quick reply.

Currently I am doing my research using ROS1, later I would definitely switch to ROS2. But now I am working on some demonstration to my supervisor and partner showing what we can do.

I would like to control the robotic arm to scan an object with constant contact force. I think it is possible with your old code. But I donot know how to set the target force. That is why I wrote to you all.

Do you know anyone who is still working on this?

Thank you very much.

ros-controls/ros2_control_demos#206<https://github.com/ros-controls/ros2_control_demos/pull/206>

gavanderhoorn, 2022-10-07:
Could I ask you to please continue this discussion somewhere else?

This is a closed issue on the tracker of a defunct ROS 1 driver for a particular robot brand.

Admittance control isn't really on-topic here.