ros-industrial-attic/ur_modern_driver #360: CMake Warning

HRjunior, 2019-10-23:
while doing the catkin build, it won't recognize hardware_interface in ur_modern_driver-master/src/ur_hardware_interface.cpp, :180::22 and 186:22

gavanderhoorn, 2019-10-23:
> in `ur_modern_driver-master/src`

you appear to have downloaded a `.zip` of the `master` branch.

As stated in the readme, `master` is deprecated and should not be used:

> ## Deprecation notice
> 
> **note**: the `master` branch contains a deprecated version of this driver, kept for archival purposes only.
> 
> The `master` version of `ur_modern_driver` is **not compatible** with newer controller and Polyscope versions, and should not be used with those.
> 
> We recommend anyone using Polyscope versions 3 and newer to use the `kinetic-devel` branch instead.

If you still want to download a `.zip`, make sure to first switch to `kinetic-devel` using the branch selector drop-down and only then download the `.zip`.

gavanderhoorn, 2019-10-23:
Closing as `wontfix`, as `master` is deprecated.

closed 2019-10-23 by gavanderhoorn