ros-industrial-attic/ur_modern_driver #361: Which version of the ur_modern_driver is the most up to date?

mlautman, 2019-11-08:
@miguelprada is the ur_modern_driver officially supported for melodic?  I noticed that some people were using https://github.com/miguelprada/ur_modern_driver/tree/ur-e with melodic. Is that branch still being actively developed or will it be merged back into master soon? (Last commit was Feb.) 

I am working on a project that involves using the UR5 and UR5-e in melodic with polyscope version > 3.7 / 5.1 and I was hoping to get some insight into which fork/branch we should be keeping an eye on.

gavanderhoorn, 2019-11-08:
> I am working on a project that involves using the UR5 and UR5-e in melodic with polyscope version > 3.7 / 5.1 and I was hoping to get some insight into which fork/branch we should be keeping an eye on.

none and neither.

You should be using [UniversalRobots/Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver).

(although technically it is a fork of `ur_modern_driver`)

gavanderhoorn, 2019-11-08:
As this is not an issue with the package in this repository, I'll close this issue.

Feel free to keep commenting on it of course.

closed 2019-11-08 by gavanderhoorn

mlautman, 2019-11-12:
Thanks @gavanderhoorn