ros-industrial-attic/ur_modern_driver #362: How does the controller re-segment the trajectory and control the motor?

xarthurx, 2019-11-20:
The question is not related to ROS but I guess it is related to how the driver is developed? I cannot think of any other places that I can ask these questions.

We're trying to add a 7th axis turntable and sync with the robot simultaneously.
Given that we only have a stepper motor and no feedback system, we're relying on the syncing on threading inside the robot controller (after testing threading from PC, the communication to UR takes longer than to the turntable).

So what we have developed is like this: 
We pre-computed a series of pose both for the robot and the turntable, and send to the two machines separately. The turntable will be put into a "waiting" state.

The scripts sent to the robot contains the joint pose plus an "on/off" signal, so that we can open two threads on the robot controller: one for controlling the motors of the robot, one for telling the turntable to rotate.

Now the question comes: based on the urscript manual, threads are executed based on a 125Hz / 0.008s time frame, and if one doesn't finish, it will be put into the queue so that the next thread will take the next 0.008 time frame.

However, since the urscript command `moveJ/moveL` is high level, there must be some mechanism inside the driver to interpret it into a higher frequency (for instance, 10kHz) signal to control the current sent to the motor, with a fixed time interval. 

What is the mechanism behind it?

Or what will the robot do if the thread controlling it is set to sleep while the other thread is sending signal to the turntable? 
(In other words, will the robot take 2x time for executing the same trajectory if using one of the two threads, compared with the case that only one thread is opened to control the robot only?)

gavanderhoorn, 2019-11-20:
> However, since the urscript command `moveJ/moveL` is high level, there must be some mechanism inside the driver to interpret it into a higher frequency (for instance, 10kHz) signal to control the current sent to the motor, with a fixed time interval.

No, there isn't.

We're just using `servoj(..)` and `speedj(..)` here.

> What is the mechanism behind it?

Unless I'm completely misunderstanding you: there is none.

We don't control the motors directly.

The driver specifies desired joint poses (and velocities) for the UR controller to reach at specific times. All the rest of the (low-level) control is done natively by the controller.

gavanderhoorn, 2019-11-20:
As this is more of a question, not an actual issue with the package itself, I'm going to close it.

Feel free to keep commenting on the issue of course,

closed 2019-11-20 by gavanderhoorn