ros-industrial-attic/ur_modern_driver #370: ursim offline simulator connect Ros in ubuntu

TFLQW, 2020-04-28:
"
Ubuntu 16.04 
URsim-3.12.1.90940
Ros Version : kinetic
"

First of all I run the ursim in one terminal the command is ./start-ursim.h
After it has been run, I execute the following command in another terminal:
roslaunch ur_modern_driver ur5_bringup.launch robot_ip:=127.0.0.1
but it occur the error:
`... logging to /home/tiboy/.ros/log/cf7e649a-8956-11ea-bbea-645a045f04b6/roslaunch-tiboy-pc-20881.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tiboy-pc:36565/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /ur_driver/base_frame: base
 * /ur_driver/max_payload: 5.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix: 
 * /ur_driver/robot_ip_address: 127.0.0.1
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/tool_frame: tool0_controller

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

auto-starting new master
process[master]: started with pid [20894]
ROS_MASTER_URI=http://127.0.0.2:11311

setting /run_id to cf7e649a-8956-11ea-bbea-645a045f04b6
process[rosout-1]: started with pid [20907]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [20923]
process[ur_driver-3]: started with pid [20925]
[FATAL] [1588081654.774828318]: ERROR on binding socket for reverse communication
[ur_driver-3] process has died [pid 20925, exit code -5, cmd /home/tiboy/UR_control/devel/lib/ur_modern_driver/ur_driver __name:=ur_driver __log:=/home/tiboy/.ros/log/cf7e649a-8956-11ea-bbea-645a045f04b6/ur_driver-3.log].
log file: /home/tiboy/.ros/log/cf7e649a-8956-11ea-bbea-645a045f04b6/ur_driver-3*.log
`

how should I solve this problem

gavanderhoorn, 2020-04-28:
First things first: `ur_modern_driver` has been deprecated, so can you confirm you actually need to use it?

The `URsim-3.12.1.90940` in your description makes me believe you have a CB3 controller.

If that is the case, please use [UniversalRobots/Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver).

TFLQW, 2020-04-29:
#[https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/165](url)

TFLQW, 2020-04-29:
I have add a new issue in # [https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/issues/165](url)

closed 2020-04-29 by TFLQW