ros-industrial-attic/ur_modern_driver #372: Driver Communication disconnected

chen128, 2020-06-03:
As the problem terminal shows below
roslaunch ur_modern_driver ur5_bringup.launch limited:=true robot_ip:=192.168.1.11
... logging to /home/cr/.ros/log/c3d11c8a-a573-11ea-98db-54ee75c732ec/roslaunch-cr-Lenovo-6645.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cr-Lenovo:33623/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /ur_driver/base_frame: base
 * /ur_driver/max_joint_difference: 0.01
 * /ur_driver/max_payload: 5.0
 * /ur_driver/max_velocity: 10.0
 * /ur_driver/max_waiting_time: 2.0
 * /ur_driver/min_payload: 0.0
 * /ur_driver/prefix: 
 * /ur_driver/require_activation: Never
 * /ur_driver/reverse_ip_address: 
 * /ur_driver/reverse_port: 50001
 * /ur_driver/robot_ip_address: 192.168.1.11
 * /ur_driver/servoj_gain: 100.0
 * /ur_driver/servoj_lookahead_time: 1.0
 * /ur_driver/servoj_time: 0.008
 * /ur_driver/servoj_time_waiting: 0.001
 * /ur_driver/shutdown_on_disconnect: True
 * /ur_driver/time_interval: 0.008
 * /ur_driver/tool_frame: tool0_controller
 * /ur_driver/use_lowbandwidth_trajectory_follower: False
 * /ur_driver/use_ros_control: False

NODES
  /
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    ur_driver (ur_modern_driver/ur_driver)

auto-starting new master
process[master]: started with pid [6658]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to c3d11c8a-a573-11ea-98db-54ee75c732ec
process[rosout-1]: started with pid [6671]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [6688]
process[ur_driver-3]: started with pid [6689]
[ INFO] [1591172723.406831507]: Setting up connection: 192.168.1.11:30003
[ INFO] [1591172723.406980852]: Connection established for 192.168.1.11:30003
[ INFO] [1591172723.407010149]: Setting up connection: 192.168.1.11:30001
[ INFO] [1591172723.407088429]: Connection established for 192.168.1.11:30001
[ INFO] [1591172723.809259999]: Got VersionMessage:
[ INFO] [1591172723.809326505]: project name: URControl
[ INFO] [1591172723.809354520]: version: 3.12.0
[ INFO] [1591172723.809387095]: build date: 12-11-2019, 15:17:47
[ INFO] [1591172723.809416126]: Disconnecting from 192.168.1.11:30001
[ INFO] [1591172723.832981992]: ActionServer enabled
[ INFO] [1591172723.833082758]: Use standard trajectory follower
[ INFO] [1591172723.838514269]: Setting up connection: :50001
[ INFO] [1591172723.839087949]: Connection established for :50001
[ INFO] [1591172723.839371183]: Initializing ur_driver/URScript subscriber
[ INFO] [1591172723.850207677]: The ur_driver/URScript initialized
[ INFO] [1591172723.850278267]: Notifier: Pipeline disconnect will shutdown the node
[ INFO] [1591172723.861647477]: Service 'ur_driver/robot_enable' activation mode: Never
[ INFO] [1591172723.861719748]: Starting main loop
[ INFO] [1591172723.861944978]: Setting up connection: 192.168.1.11:30003
[ INFO] [1591172723.862074550]: Starting pipeline RTPacket
[ INFO] [1591172723.862244930]: Connection established for 192.168.1.11:30003
[ INFO] [1591172723.862425480]: Setting up connection: 192.168.1.11:30002
[ INFO] [1591172723.862517665]: Starting pipeline StatePacket
[ INFO] [1591172723.862678007]: Connection established for 192.168.1.11:30002
[ INFO] [1591172723.866123744]: Starting ActionServer
[ INFO] [1591172723.866371738]: Trajectory thread started
[ERROR] [1591172723.923255137]: Sub-package of type 3 was not parsed completely!
[ INFO] [1591172723.923366359]: Disconnecting from 192.168.1.11:30002
[ INFO] [1591172723.923455835]: Shutting down on stopped pipeline StatePacket
[ur_driver-3] process has died [pid 6689, exit code -11, cmd /home/cr/jt_xuanwu/devel/lib/ur_modern_driver/ur_driver __name:=ur_driver __log:=/home/cr/.ros/log/c3d11c8a-a573-11ea-98db-54ee75c732ec/ur_driver-3.log].
log file: /home/cr/.ros/log/c3d11c8a-a573-11ea-98db-54ee75c732ec/ur_driver-3*.log

I'm sure about my pc and ur5 ip address ,and it can ping successfully.
Before it occured i have controled UR5,one day this error message just happened.I try to reload all the univeral robot package,but it still happened.I also google it.Maybe no one meet it before.
Thanks for your help.

gavanderhoorn, 2020-06-03:
What type of controller are you using? Is it a CB3?

chen128, 2020-06-04:
yes，it is CB3.The new vision software.

gavanderhoorn, 2020-06-04:
Then you should not be using `ur_modern_driver` as stated in [the readme](https://github.com/ros-industrial/ur_modern_driver/blob/066716ed6a10eb0d9211f5bc5e32d65c7afce878/README.md#deprecation-notice) and in multiple other places.

Use [UniversalRobots/Universal_Robots_ROS_Driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver).

`ur_modern_driver` will not work with CB3 controllers any more.

closed 2020-06-04 by gavanderhoorn