ros-industrial-attic/ur_modern_driver #374: Assertion failed due to BinParser

ipa-mjp, 2020-06-24:
Hello,

I am controlling UR10 robot using ```ros_control```. While launching ```ur_modern_driver``` with ros component, I am getting following error from BinParser.
```
.../ur_modern_driver/bin_parser.h:89: T BinParser::peek() [with T = long unsigned int]: Assertion `buf_pos_ + sizeof(T) <= buf_end_' failed.
```

I run my urdf and controller configuration in the simulation and it works fine without having any error (```universal_robot``` package). I used same configuration for ```ur_modern_driver``` and could not able to launch successfully. 

I am using kinetic as ros_distro and same branches from ```ur_modern_driver``` and ```universal_robot```.

Is there anything wrong, I configured? or Is anyone facing same problem and solve it?

  

Thank you.

gavanderhoorn, 2020-06-24:
Please tell us which controller series you have.

CB2, CB3 or e-Series?

ipa-mjp, 2020-06-24:
Hello,

Thank you for prompt reply. I am using CB2 with UR10 robot.

gavanderhoorn, 2020-06-24:
Which system software version?

ipa-mjp, 2020-07-03:
Hello,
That was my mistake as I build workspace with ```catkin build``` instead of ```catkin_make``` as mention on README.md. Is this a really issue?

gavanderhoorn, 2020-07-03:
> Hello,
> That was my mistake as I build workspace with `catkin build` instead of `catkin_make` as mention on README.md. Is this a really issue?

No.

That should not have any effect.

bierm4nn, 2020-11-20:
Hello, 
I am facing the same Issue. I am using a UR5 with a CB1 and Polyscope 1.5.7849 with Ubuntu 16.04 and ros-kinetic.

Is it possible that this has sth to do with ur_msgs, as this package is now included in the new ur_robot_driver and a dependency of ur_modern_driver?

Do you have a Idea where the problem could be?
@ipa-mjp: Did you solve your problem?

Thanks for your help <3