ros-industrial-attic/ur_modern_driver #375: /ur_driver/URScript speedj command can not reach the desired velocity

Eugene19950, 2020-07-03:
I am use "rostopic pub /ur_driver/URScript std_msgs/String "speedj([0.3,0.2,0.1,0,0,0], 0.5, 1.5)"
  pub a velocity to ur5, but when I listen to the  feedback /joint_states/velocity of the robot, I find that the joint velocity doesn't reach the desired one, as seen below
[image]
but if I just command joint speed to only one joint by  "rostopic pub /ur_driver/URScript std_msgs/String "speedj([0.3,0,0,0,0,0], 5, 1)",(i set acc to 5 and time to 1s), the velocity is reached, anyone knows the reason?
[image]

gavanderhoorn, 2020-07-03:
Please tell us which controller you are using.

If you're using a CB3 or e-Series robot, we can't support you here, as this driver is deprecated.

Eugene19950, 2020-07-04:
I'm using the CB3 controller but the software version remains 3.4, so I choose ur_moder_driver to drive my ur5 robot.

gavanderhoorn, 2020-07-04:
I'm sorry, but for anything CB3 this driver has been deprecated.

This means we won't field any support questions of users using `ur_modern_driver` with CB3 controllers.

I'm going to close this as a `wontfix` and would really suggest you upgrade to a later version and use `ur_robot_driver`.

closed 2020-07-04 by gavanderhoorn