ros-industrial-attic/ur_modern_driver #377: Robot Move inaccuracy when it move fast

elevenjiang1, 2020-12-25:
When I let robot move slow (0.1m/s),the robot will move accurcy(max 3 mm inaccuracy ). However , when I let robot move fast (0.3m/s) , the robot will move inaccuracy (max 3cm inaccuracy ) !
I guess the reason it that I give wrong JointTrajectoryPoint().time_from_start() . The client.wait_for_result() will finish but robot is still moving , this is the reason why I think the robot move inaccuracy .

My code is below :
```python
path_length=distance(from_pose,target_pose)
t=path_length/vel #vel=0.4
sample_number=t*sample_freq  #sample_freq=10
interpolation_pose=interpolation(from_pose,target_pose,sample_number)
joints_all=inverse_kinematic(interpolation_pose)

temp_time=t/joints_all.shape[0]
goal=FollowJointTrajectoryGoal()
for i in range(joints_all.shape[0]):
      q=JointTrajectoryPoint()
      joint_list=joints_all[i,:].tolist()
      q.positions=joint_list
      q.velocities=[0.]*6
      q.time_from_start=rospy.Duration(temp_time*(i+1))
      goal.trajectory.points.append(q)

client.send(goal)
client.wait_for_result()
```

gavanderhoorn, 2020-12-25:
Please first tell us what version of Polyscope your UR controller is running.

Is it a CB3 or a CB2?

elevenjiang1, 2020-12-25:
CB3 (but can i use modern_driver but not [ur_robot_driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) )

gavanderhoorn, 2020-12-26:
Anyone with a CB3 should use `ur_robot_driver`.

`ur_modern_driver` is deprecated for CB3s and newer.

Unfortunately we'll not support you with `ur_modern_driver` with a CB3.

You should be able to upgrade the software on the controller to the minimum version required for `ur_robot_driver`.

gavanderhoorn, 2020-12-26:
With that I'm going to close this issue.

Feel free to keep commenting on it of course.

closed 2020-12-26 by gavanderhoorn