ros-industrial-attic/ur_modern_driver #378: Received a goal with incorrect joint names

robint-XNF, 2021-04-22:
when i use ssh and launch  :
`roslaunch ur_modern_driver ur3_hande_bringup.launch robot_ip:=192.168.10.100`
then i use moveit to control it,and send a goal and click "excute" in the gui,then i got an error:
Received a goal with incorrect joint names: elbow_joint shoulder_lift_joint shoulder_pan_joint wrist_1_joint wrist_2_joint wrist_3_joint
anyone can help me?

gavanderhoorn, 2021-04-22:
What sort of controller do you have?

CB3?

robint-XNF, 2021-04-22:
> What sort of controller do you have?
> 
> CB3?

ur3,moveit ,

gavanderhoorn, 2021-04-22:
What is the type of the Universal Robot controller box?

robint-XNF, 2021-04-22:
> What is the type of the Universal Robot controller box?
> 
> CB3 or e-Series?

CB3

gavanderhoorn, 2021-04-22:
I'm going to close this, as, as noted in the readme, CB3 and newer users should be using the official UR driver:

> ## Deprecation notice
> 
> **note**: the `master` branch contains a deprecated version of this driver, kept for archival purposes only.
> 
> `ur_modern_driver` has been deprecated completely in favour of [ur_robot_driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver).
> 
> Users with CB3 and e-Series controllers should use `ur_robot_driver`.
> 
> Refer to the [announcement on ROS Discourse](https://discourse.ros.org/t/deprecation-of-ur-modern-driver/11629) for more information.
>

closed 2021-04-22 by gavanderhoorn