ros-industrial-attic/ur_modern_driver #379: UR5 not moving after receiving trajectory

angux94, 2021-06-08:
Hello, I'm using MoveIt along with an UR5, I'm running at the moment:

- roslaunch ur_modern_driver ur5_bringup.launch
- roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch
- roslaunch ur5_moveit_config moveit_rviz.launch config:=true

And it seems to be working successfully and connecting/communicating to the robot, but when I send some basic planning through "Plan and execute" in Rviz, the UR5 seems to get the request but it never moves.

I'm on **Polyscope 3.6** using **ROS melodic**. (I haven't used [https://github.com/UniversalRobots/Universal_Robots_ROS_Driver](url) as it says it supports from CB3 >= 3.7).

From my terminal outputs i'm getting:

**Terminal 1: roslaunch ur_modern_driver ur5_bringup.launch**
```
[ INFO] [1623140863.575058100]: Setting up connection: 192.168.0.11:30003
[ INFO] [1623140863.579746600]: Connection established for 192.168.0.11:30003
[ INFO] [1623140863.580503700]: Setting up connection: 192.168.0.11:30001
[ INFO] [1623140863.584793000]: Connection established for 192.168.0.11:30001
[ INFO] [1623140863.739644200]: Got VersionMessage:
[ INFO] [1623140863.739708300]: project name: URControl
[ INFO] [1623140863.739724000]: version: 3.6.0
[ INFO] [1623140863.739734800]: build date: 14-05-2018, 16:43:28
[ INFO] [1623140863.739750200]: Disconnecting from 192.168.0.11:30001
[ INFO] [1623140863.775175500]: ActionServer enabled
[ INFO] [1623140863.777238300]: Use standard trajectory follower
[ INFO] [1623140863.779432000]: Setting up connection: :50001
[ INFO] [1623140863.779639500]: Connection established for :50001
[ INFO] [1623140863.780836700]: Initializing ur_driver/URScript subscriber
[ INFO] [1623140863.785054600]: The ur_driver/URScript initialized
[ INFO] [1623140863.785096900]: Notifier: Pipeline disconnect will shutdown the node
[ INFO] [1623140863.794686800]: Service 'ur_driver/robot_enable' activation mode: Never
[ INFO] [1623140863.794781800]: Starting main loop
[ INFO] [1623140863.795025900]: Starting pipeline RTPacket
[ INFO] [1623140863.795226700]: Setting up connection: 192.168.0.11:30003
[ INFO] [1623140863.795451500]: Starting pipeline StatePacket
[ INFO] [1623140863.796534600]: Setting up connection: 192.168.0.11:30002
[ INFO] [1623140863.803307900]: Starting ActionServer
[ INFO] [1623140863.803468600]: Connection established for 192.168.0.11:30003
[ INFO] [1623140863.808018200]: Trajectory thread started
[ INFO] [1623140863.810610300]: Connection established for 192.168.0.11:30002
[ INFO] [1623140929.396165100]: Received new goal
[ INFO] [1623140929.396353400]: Trajectory received and accepted
[ INFO] [1623140929.396622500]: Translating trajectory
[ INFO] [1623140929.396782100]: Executing trajectory with 4 points and duration of 0.916s
[ INFO] [1623140929.396939100]: Attempting to start follower 0x55af124e6ae0
[ INFO] [1623140929.397122900]: Uploading trajectory program to robot

```

**Terminal 2: roslaunch ur5_moveit_config ur5_moveit_planning_execution.launch**
```
[ INFO] [1623140879.241233100]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1623140879.241320200]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1623140929.280514700]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1623140929.293039000]: Planning attempt 1 of at most 1
[ INFO] [1623140929.322197000]: Planner configuration 'manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1623140929.329240300]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623140929.330379200]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623140929.330667200]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623140929.332183700]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623140929.336746600]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1623140929.344971100]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1623140929.348419400]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1623140929.350581200]: RRTConnect: Created 5 states (3 start + 2 goal)
[ INFO] [1623140929.351303100]: ParallelPlan::solve(): Solution found by one or more threads in 0.023427 seconds
[ INFO] [1623140929.352447400]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623140929.353083000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623140929.355397600]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1623140929.355537600]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1623140929.355928000]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623140929.356553700]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623140929.359327000]: RRTConnect: Created 4 states (2 start + 2 goal)
[ INFO] [1623140929.360967300]: RRTConnect: Created 5 states (3 start + 2 goal)
[ INFO] [1623140929.361241400]: ParallelPlan::solve(): Solution found by one or more threads in 0.009392 seconds
[ INFO] [1623140929.361603500]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623140929.364021700]: RRTConnect: Created 5 states (2 start + 3 goal)
[ INFO] [1623140929.364376800]: RRTConnect: Starting planning with 1 states already in datastructure
[ INFO] [1623140929.367924200]: RRTConnect: Created 5 states (3 start + 2 goal)
[ INFO] [1623140929.368262300]: ParallelPlan::solve(): Solution found by one or more threads in 0.006804 seconds
[ INFO] [1623140929.380289300]: SimpleSetup: Path simplification took 0.011250 seconds and changed from 3 to 2 states
[ INFO] [1623140929.388854300]: Disabling trajectory recording

```
**Terminal 3: roslaunch ur5_moveit_config moveit_rviz.launch config:=true**
```
[ INFO] [1623140888.318860700]: rviz version 1.13.16
[ INFO] [1623140888.319363600]: compiled against Qt version 5.9.5
[ INFO] [1623140888.319407500]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1623140888.335579300]: Forcing OpenGl version 0.
[ INFO] [1623140888.926196700]: Stereo is NOT SUPPORTED
[ INFO] [1623140888.926361300]: OpenGL device: llvmpipe (LLVM 8.0, 256 bits)
[ INFO] [1623140888.926444900]: OpenGl version: 3.1 (GLSL 1.4).
[ INFO] [1623140892.877808400]: Loading robot model 'ur5'...
[ WARN] [1623140892.877986100]: Skipping virtual joint 'fixed_base' because its child frame 'base_link' does not match the URDF frame 'world'
[ INFO] [1623140892.878051600]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1623140892.926370800]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/rviz_fc5ae47b5a94_402_6280233072355241255/manipulator/kinematics_solver_attempts' from your configuration.
[ INFO] [1623140893.408089700]: Starting planning scene monitor
[ INFO] [1623140893.413558900]: Listening to '/move_group/monitored_planning_scene'
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
QObject::connect: Cannot queue arguments of type 'QVector<int>'
(Make sure 'QVector<int>' is registered using qRegisterMetaType().)
[ INFO] [1623140894.027250300]: Constructing new MoveGroup connection for group 'manipulator' in namespace ''
[ INFO] [1623140895.639155800]: Ready to take commands for planning group manipulator.
[ INFO] [1623140895.643379300]: Looking around: no
[ INFO] [1623140895.643670600]: Replanning: no
[ INFO] [1623140895.704513200]: Marker 'EE:goal_ee_link/10': mesh_use_embedded_materials is ignored.

```
I can't identify the problem as it seems to be working, any idea on what might be the issue?

Thanks in advance.

gavanderhoorn, 2021-06-08:
> I'm on **Polyscope 3.6** using **ROS melodic**. (I haven't used [https://github.com/UniversalRobots/Universal_Robots_ROS_Driver](url) as it says it supports from CB3 >= 3.7).

CB3 and newer controllers should really use `Universal_Robots_ROS_Driver`.

`ur_modern_driver` is deprecated for everything CB3 and up.

gavanderhoorn, 2021-06-08:
Due to what I wrote in my earlier comment I'm going to close this.

I'd really recommend you upgrade your Polyscope version to `>=3.7` and use `Universal_Robots_ROS_Driver` instead of `ur_modern_driver`.

`Universal_Robots_ROS_Driver` is supported by Universal Robots, supports more features and is actively being developed and maintained.

closed 2021-06-08 by gavanderhoorn