ros-industrial-attic/ur_modern_driver #380: No transform available between frame 'tool0_controller' and planning frame '/world' ()

baifanxxx, 2021-09-16:
Hi,
I have a problem about UR3 on compute Cartesian path.

my configration is:
Ubuntu 16.04, ROS kinetic, UR3, ros-industrial/universal_robot + ur_modern_driver.

If I use
(plan, fraction) = group.compute_cartesian_path( waypoints,0.01, 0.0, True)
to compute Cartesian path. There are some erro like this:

[ INFO] [1631799290.770572209]: Received request to compute Cartesian path
[ WARN] [1631799290.770594171]: No transform available between frame 'tool0_controller' and planning frame '/world' ()
[ERROR] [1631799290.770637100]: TF Problem: "world" passed to lookupTransform argument target_frame does not exist.
[ERROR] [1631799290.770649077]: Error encountered transforming waypoints to frame '/world'

I think it maybe something wrong on UR3 URDF model. However, I can successfully use

  group.go(joint_goal, wait=True)
and

group.set_start_state_to_current_state()
group.set_pose_target(target_pose, end_effector_link)
traj = group.plan()
group.execute(traj)
rospy.sleep(1)
group.go()
This is my config:

rospy.init_node('ipm_detection',anonymous=True)
moveit_commander.roscpp_initialize(sys.argv)
group = moveit_commander.MoveGroupCommander('manipulator')

# Create a goal to send (to the action server)
goal = control_msgs.msg.GripperCommandGoal()

robot_commander = moveit_commander.RobotCommander()

end_effector_link = group.get_end_effector_link()
reference_frame = 'base_link'
group.set_pose_reference_frame(reference_frame)

group.allow_replanning(True)

group.set_goal_position_tolerance(0.0001)
group.set_goal_orientation_tolerance(0.001)
group.set_max_acceleration_scaling_factor(0.05)
group.set_max_velocity_scaling_factor(0.05)
I guess this may not be my own problem, but the urdf file in the package or the controller. So I hope everyone or official maintainers to help me. The matter is urgent, and I desperately need your help. grateful.

Thank you!

gavanderhoorn, 2021-09-16:
> Ubuntu 16.04, ROS kinetic, UR3, ros-industrial/universal_robot + ur_modern_driver.

Unfortunately we don't support UR3s + `ur_modern_driver`.

You should really use `UniversalRobots/Universal_Robots_ROS_Driver` for that robot.

closed 2021-09-16 by gavanderhoorn

baifanxxx, 2021-09-16:
Sorry for that. 
But I can only use UR3 on ubuntu 18? It is a pity that I have config all my dependency on Ubuntu16. The new package  UniversalRobots/Universal_Robots_ROS_Driver seem support >unbuntu18 only?

gavanderhoorn, 2021-09-16:
ROS Kinetic has been end-of-life since April 2021. You're using an EOL version of ROS, on a nearly EOL version of Ubuntu with an EOL package (`ur_modern_driver` has been EOL for CB3 and e-Series controllers since Nov 2019).

gavanderhoorn, 2021-09-16:
But this seems to be a MoveIt problem. Are you also using Kinetic's version of MoveIt? There have been massive updates and fixes to MoveIt since the last Kinetic release of that package, which you're missing out on.

baifanxxx, 2021-09-16:
Thank you, I have updated my MoveIt but not work. I will continue to solve this problem on Ubuntu16 to prevent the configuration of a large number of environment dependencies.