ros-industrial-attic/ur_modern_driver #382: [ERROR] [1659339804.301066322]: Ignoring transform for child_frame_id "tool0_controller" from authority "unknown_publisher" because of an invalid quaternion in the transform (-nan -nan -nan -nan)

ozilxu, 2022-08-01:
[image]
Hi! 
I got this error when launching the moveit launch file. And I have added the tool0_controller setting into the ur5.urdf.xacro file

[image]
May I ask how to fix this error? @jeppewalther @NikolasE @fmauch @NikolasE @fmauch @ipa-nhg @miguelprada thanks!!

fmauch, 2022-08-01:
You'll also need a hardware instance running in order to fix this. Please note that the `ur_modern_driver` is considered deprecated. Please use the [`ur_robot_driver`](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver) instead.

The transform in question will be published by the driver directly.

gavanderhoorn, 2022-08-01:
@ozilxu: could I ask you to please not spam @-mention all those people?

Some have not been involved with this repository for a long time, and it's not very nice to ping them directly.

gavanderhoorn, 2022-08-01:
As @fmauch already wrote: unless you have a CB1 or CB2 controller, you should not be using this package.

This is clearly stated [in the readme](https://github.com/ros-industrial/ur_modern_driver/tree/066716ed6a10eb0d9211f5bc5e32d65c7afce878#deprecation-notice).

For now I'll close this, and assume you have CB3 or e-Series controller.

closed 2022-08-01 by gavanderhoorn