ros-industrial-attic/ur_modern_driver #383: error while trying to install

ilariagi, 2022-11-10:
following the installation guide, on a virtual machine with ubuntu 18.04.6 and ros melodic
rosdep install --from-paths src --ignore-src --rosdistro melodic

ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies:
ur_modern_driver: Cannot locate rosdep definition for [ur_description]

fmauch, 2022-11-10:
First of all the obligatory question: Are you sure you want to use this driver? The answer to this should only be yes, if you have a CB1 or CB2 robot as stated in the README. If you have a CB3 or e-series robot, please use the [ur_robot_driver](https://github.com/UniversalRobots/Universal_Robots_ROS_Driver).

That being said, currently it is required to build ur_description from source, please checkout the [universal_robot](https://github.com/ros-industrial/universal_robot) repo on the `melodic-devel` branch. A binary release will be available very soon.

Please also note, that the launchfile to load the description has changed, so the launchfile's name will have to be changed ,e.g. to `load_ur5.launch` in this line:

https://github.com/ros-industrial/ur_modern_driver/blob/066716ed6a10eb0d9211f5bc5e32d65c7afce878/launch/ur5_ros_control.launch#L18

ilariagi, 2022-11-11:
Yes, we have to use a ur5 cb2 with polyscope v1.8.x.

Thank you

ilariagi, 2022-11-11:
I checked out the universal_robot repo on the correct branch, and I can build it alone successfully, but when I try to build the workspace including also ur_modern_driver it fails in ros_main.cpp.o

/home/ilaria/ros_workspace/src/ur_modern_driver/include/ur_modern_driver/ros/io_service.h: In member function ‘bool IOService::setPayload(ur_msgs::SetPayloadRequest&, ur_msgs::SetPayloadResponse&)’:
/home/ilaria/ros_workspace/src/ur_modern_driver/include/ur_modern_driver/ros/io_service.h:69:54: error: ‘ur_msgs::SetPayloadRequest {aka struct ur_msgs::SetPayloadRequest_<std::allocator<void> >}’ has no member named ‘payload’
     return (resp.success = commander_.setPayload(req.payload));
                                                      ^~~~~~~
ur_modern_driver/CMakeFiles/ur_driver.dir/build.make:422: recipe for target 'ur_modern_driver/CMakeFiles/ur_driver.dir/src/ros_main.cpp.o' failed
make[2]: *** [ur_modern_driver/CMakeFiles/ur_driver.dir/src/ros_main.cpp.o] Error 1
CMakeFiles/Makefile2:5865: recipe for target 'ur_modern_driver/CMakeFiles/ur_driver.dir/all' failed
make[1]: *** [ur_modern_driver/CMakeFiles/ur_driver.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

closed 2024-07-02 by gavanderhoorn