ros-industrial-attic/ur_modern_driver #386: UR10 CB2 can't connect with controller correctly

ChenZhenY, 2023-10-12:
Hi, I am controlling a CB2 UR10 arm using ur_modern_driver. I have verified the Eternet connection and run `test_move.py` in the driver to make sure the joint controller can control the robot arm. However, when I try to run the Moveit, the robot doesn't move. Here are my error messages with and without ros_control.

Environment: Ubuntu20.04+ROS Noetic+latest ur_modern_driver

### Using follow_joint_trajectory controller
I can't make Moveit talk to the controller correctly.
1. ` roslaunch ur_modern_driver ur10_bringup.launch robot_ip:=192.168.1.100` (this works for test_move.py, indicating low-level communication is good)
2. `ros_controller.yaml` (I think the original one is for ros_control?)
```
controller_list:
- name: follow_joint_trajectory
  action_ns: follow_joint_trajectory
  type: FollowJointTrajectory
  joints:
    - shoulder_pan_joint
    - shoulder_lift_joint
    - elbow_joint
    - wrist_1_joint
    - wrist_2_joint
    - wrist_3_joint
```
3. ` roslaunch ur10_moveit_config moveit_planning_execution.launch` with error `Action client not connected: follow_joint_trajectory/follow_joint_trajectory`.
```
... logging to /home/czy/.ros/log/78372c38-67b7-11ee-85d0-fd59ebf7c67c/roslaunch-andromeda-955123.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://andromeda:42909/

SUMMARY
========

PARAMETERS
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities:
 * /move_group/controller_list: [{'name': 'follow...
 * /move_group/disable_capabilities:
 * /move_group/jiggle_fraction: 0.05
 * /move_group/manipulator/default_planner_config: RRTConnect
 * /move_group/manipulator/longest_valid_segment_fraction: 0.01
 * /move_group/manipulator/planner_configs: ['SBL', 'EST', 'L...
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BFMT/balanced: 0
 * /move_group/planner_configs/BFMT/cache_cc: 1
 * /move_group/planner_configs/BFMT/extended_fmt: 1
 * /move_group/planner_configs/BFMT/heuristics: 1
 * /move_group/planner_configs/BFMT/nearest_k: 1
 * /move_group/planner_configs/BFMT/num_samples: 1000
 * /move_group/planner_configs/BFMT/optimality: 1
 * /move_group/planner_configs/BFMT/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMT/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECE/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECE/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECE/range: 0.0
 * /move_group/planner_configs/BKPIECE/type: geometric::BKPIECE
 * /move_group/planner_configs/BiEST/range: 0.0
 * /move_group/planner_configs/BiEST/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRT/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRT/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRT/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRT/init_temperature: 100
 * /move_group/planner_configs/BiTRRT/range: 0.0
 * /move_group/planner_configs/BiTRRT/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRT/type: geometric::BiTRRT
[... 441 lines cut ...]

[ INFO] [1696974323.340785299]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1696974323.340793645]: MoveGroup context initialization complete

You can start planning now!

[pub_text-1] process has died [pid 949661, exit code 1, cmd /opt/ros/noetic/lib/rostopic/rostopic pub /msg std_msgs/String Data: Please keep this running in a separate tab. __name:=pub_text __log:=/home/czy/.ros/log/3eb6bb82-67b6-11ee-85d0-fd59ebf7c67c/pub_text-1.log].
log file: /home/czy/.ros/log/3eb6bb82-67b6-11ee-85d0-fd59ebf7c67c/pub_text-1*.log
[ INFO] [1696974345.462834967]: Combined planning and execution request received for MoveGroup action. Forwarding to planning and execution pipeline.
[ INFO] [1696974345.462948738]: Planning attempt 1 of at most 1
[ INFO] [1696974345.465977008]: Planner configuration 'manipulator[RRTConnect]' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1696974345.467110061]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1696974345.467231074]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1696974345.467376479]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1696974345.467491586]: manipulator/manipulator[RRTConnect]: Starting planning with 1 states already in datastructure
[ INFO] [1696974345.478479169]: manipulator/manipulator[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1696974345.478992162]: manipulator/manipulator[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1696974345.479086460]: manipulator/manipulator[RRTConnect]: Created 4 states (2 start + 2 goal)
[ INFO] [1696974345.479287718]: manipulator/manipulator[RRTConnect]: Created 5 states (2 start + 3 goal)
[ INFO] [1696974345.479512673]: ParallelPlan::solve(): Solution found by one or more threads in 0.012940 seconds
[ INFO] [1696974345.484893959]: SimpleSetup: Path simplification took 0.005230 seconds and changed from 3 to 2 states
[ WARN] [1696974345.831903105]: Controller 'vel_based_pos_traj_controller' failed with error PATH_TOLERANCE_VIOLATED: shoulder_pan_joint path error 0.100292
[ WARN] [1696974345.832103385]: Controller handle vel_based_pos_traj_controller reports status ABORTED
[ INFO] [1696974345.832202289]: Completed trajectory execution with status ABORTED ...
```

fmauch, 2023-10-12:
I transferred this issue to the `ur_modern_driver`'s repo, as it is related to that. When you say "latest ur_modern_driver", do you mean the `master` branch or `kinetic-devel`?

`kinetic-devel` is actually newer than the `master` version. On `kinetic-devel` you can set [`use_ros_control`](https://github.com/ros-industrial/ur_modern_driver/blob/0331cbf8452e95bd1921449523468ed6fb97f3dd/launch/ur_common.launch#L17) in order to use `ros_control`. However, to make ith work together with the example  MoveIt! packages, you probably will have to modify its [controller config](https://github.com/ros-industrial/universal_robot/blob/melodic-devel/ur10_moveit_config/config/ros_controllers.yaml) to this:

```yaml
controller_list:
- name: "pos_based_pos_traj_controller"
  action_ns: follow_joint_trajectory
  type: FollowJointTrajectory
  joints:
    - shoulder_pan_joint
    - shoulder_lift_joint
    - elbow_joint
    - wrist_1_joint
    - wrist_2_joint
    - wrist_3_joint
```

I hope, this helps you getting running.

ChenZhenY, 2023-10-12:
Thanks so much. After changing to the `kinetic-devel` and fixing some minor issues:
1. `import roslib; roslib.load_manifest('ur_modern_driver')` in `test_move.py`
2. ur10_ros_control.launch
```
  <include file="$(find ur_description)/launch/load_ur10.launch">
    <!-- <arg name="limited" value="$(arg limited)"/> -->
  </include>
```
I am able to run the Moveit and control the hardware robot arm. Do you want to consider changing the default branch of the repo to `kinetic-devel`?

closed 2023-10-12 by ChenZhenY